leggedrobotics/grand_tour_box #1166: Camera settings

Yanhof, 2025-09-29:
Hi @JonasFrey96,
Great work. I got an intro into how to use Boxi from Turcan last Friday as the plan is to gather a lunar dataset for my thesis and maybe also to add to the grand_tour_dataset. There are still some open points which might also be of interest for you which I came across:

1. Is there any reason why you set the rate if the alphasense IMU to 200 instead of 400Hz?
2. Is it possible to change the output_encoding of the HDR camera (and potentially also the Zed2i) directly to mono8 to have the images monochrome?
3. How do you know that the exposure time of the HDR was 11ms, I did not see anything in the code which had this? https://tier4.github.io/camera_docs/t4cam-ctrl/ mentions that the camera can support 1-260ms exposure time as well as auto exposure. In https://github.com/leggedrobotics/hdr_camera_driver/tree/eb8fe0b8fcdb6ad5cea0bc70272563ae55e870d2 did I not find anything about this
		-> Is there a way to tell for the alphasense images which exposure was choosen (by the auto exposure algorithm), and if AE is possible for teh HDR also for it?

Remark (probably you used different settings during the test than in the code here): In your paper do mention that you run the HDR camera with 10Hz but in the code it lists 30 (https://github.com/leggedrobotics/grand_tour_box/blob/main/box_bringup/bringup_hdr/usb_cam/launch/usb_cam_front.launch)

Thanks

Yanhof, 2025-09-30:
Hi @JonasFrey96 
Thanks.
Regarding Point 3: So, what you say is that in the test we should not use autoexposure for the C1 if we want a dataset to test VIO on? For selecting a good exposure time of the C1 for the test could (and should) we test it when we are there or what would be your recommendation? How long does the flashing take and can you point us to the file where we can change this.

JonasFrey96, 2025-09-30:
Point 3 simply use 11ms fixed exposure time - this worked great

closed 2025-10-02 by Yanhof